matthijskooijman/arduino-lmic #243: Unable to compile lmic lib with Arduino AVR Boards 1.8.1

vlastahajek, 2019-10-03:
After I've upgraded Arduino IDE to 1.8.10, which in turn brought Arduino AVR Boards 1.8.1, I'm unable to compile basic ttn-abp sketch, which uses lmic libary:
```
C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `radio_irq_handler':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/radio.c:801: undefined reference to `table_get_u2'

C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `schedRx12':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:1336: undefined reference to `table_get_ostime'

C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `calcBcnRxWindowFromMillis':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:419: undefined reference to `table_get_ostime'

C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `calcRxWindow':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:405: undefined reference to `table_get_ostime'

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:407: undefined reference to `table_get_ostime'

C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `initDefaultChannels':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:553: undefined reference to `table_get_u4'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "lmic.h"
 Used: C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic
Multiple libraries were found for "SPI.h"
 Used: C:\software\arduino.1.8.10\hardware\arduino\avr\libraries\SPI
exit status 1
Error compiling for board Arduino Pro or Pro Mini.
```
Works well again after I had downgraded back to 1.6.23

galagaking, 2019-10-07:
Workaround: create a platform.local.txt
[platform.local.txt](https://github.com/matthijskooijman/arduino-lmic/files/3696862/platform.local.txt)
 file in the <arduino_install_dir>/hardware/arduino/avr/ directory (same dir as platform.txt). File will replace -Os optimisation by -O2 and that will compile the code without any errors.

paulvha, 2019-10-10:
Or fix the real issue:
In oslmic.h, the lmic directory around line 231 it states:

	 inline type table_get ## postfix(const type *table, size_t index) { \

add **static** to beginning of line:

	static inline type table_get ## postfix(const type *table, size_t index) { \

You can now compile without a problem.

paulvha, 2019-10-10:
inline keyword is a hint to the compiler to optimize, do not call it as a the function but add into the code, but is it really used by the compiler? Some say that inline function in C++ should be used instead of macros. Inline functions are parsed by the compiler and the macros in a program are expanded by preprocessor.  It looks with the -O2 optimizer instead of the -Os it already reacts different less optimizing the code size, maybe already discarding the inline hint.  Understanding compilers in depth is a science by its own, which I do not master.

matthijskooijman, 2019-10-10:
> inline keyword is a hint to the compiler to optimize, do not call it as a the function but add into the code, but is it really used by the compiler? Some say that inline function in C++ should be [...]

As I said, the `inline` keyword is more an instruction for the linker to allow defining a function in a header file (which is historically (without LTO) needed  to allow inlining functions) without getting linker errors. The keyword might also serve as a hint to the compiler that you are expecting the function to be inlined, but in practice I think compilers simply see if it is efficient or not. If you really want to force inlining, there's also an `always_inline` attribute, but that's a completely separate (and non-portable, I think) thing.

closed 2020-01-26 by matthijskooijman

jbrepogmailcom, 2020-01-27:
Hello I understand this has been fixed in the library as per "matthijskooijman added a commit". However, when I just downloaded the library and tried to compile for esp8266 and for esp32, I still get these errors:

/home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework/src/hal/hal.cpp: In function 'void hal_printf_init()':
/home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework/src/hal/hal.cpp:223:54: error: '_FDEV_SETUP_WRITE' was not declared in this scope
     fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);
                                                      ^
/home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework/src/hal/hal.cpp:223:71: error: 'fdev_setup_stream' was not declared in this scope
     fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);
                                                                       ^
Multiple libraries were found for "lmic.h"
 Used: /home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework
Multiple libraries were found for "SPI.h"
 Used: /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SPI
Using library IBM_LMIC_framework at version 1.5.0 arduino-0 in folder: /home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework 
Using library SPI at version 1.0 in folder: /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SPI 
exit status 1
Error compiling for board Generic ESP8266 Module.

matthijskooijman, 2020-01-27:
You're using an older version:

> Using library IBM_LMIC_framework at version 1.5.0 arduino-0 in folder: /home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework

The latest is 1.5.0-arduino-3, but I just remembered that the Arduino IDE mis-sorts these version numbers, so it probably installs the wrong version by default...

commit b0f62ddc (2020-01-26): Add inline extern definitions of table_get functions

These functions are declared in oslmic.h as inline functions, with a
body to allow inlining. In C++, when a function is used but not inlined
everywhere, it is output into every compilation unit where it is used
like that, with the linker then discarding all but one definition.

However, in C, a definition is only output when a function is declared
as extern inline. In practice, this means the function body stays in the
header file, but an additional extern inline declaration is inserted
into a single source file to output the function defin [... cut]