matthijskooijman/arduino-lmic #244: compatibility with SX126X?

josjimenezja, 2019-10-04:
Hi @matthijskooijman ist possible that this lib will be work with the SX1262 from semtech?
Im using a ESP32 and E22-900M22S with SX1262

matthijskooijman, 2019-10-05:
The BasicMac stack, which is a continuation of the LMIC stack, supports the SX1262. I'm planning to port that one to Arduino as well. An initial attempt is here https://github.com/Ideetron/Basicmac-Arduino, but it's not actually working yet (and I haven't taken much time to debug yet).

RAKWireless, 2020-05-14:
Hi @matthijskooijman ,
I want to use your Basicmac Arduino LoRaWAN stack https://github.com/Ideetron/Basicmac-Arduino on my device which is nRF52840 + SX1262, but you said this repo is not actually working yet. Can i know what's the main issues and do you have a schedule about a final version repo?
Thank you!

matthijskooijman, 2020-05-14:
I continued working on BasicMac at https://github.com/LacunaSpace/basicmac, that version *is* working with SX1262 now. Check out that repo, the README should be current.

Given that I am not planning to port back the SX1262 support to this LMIC repo, I think I'm going to close this issue as a wontfix.

closed 2020-05-14 by matthijskooijman