matthijskooijman/arduino-lmic #245: ABP serial print garbage (arduino UNO)

v3yron, 2019-10-09:
Hello,

I am having garbage printed on the serial monitor of arduino.
I am using arduino UNO and Dragino shield v1.4 and i am using ABP activation method.
The problem is that while i am receiving my messages on the gateway and it works fine but the serial monitor doesn't show anything at all it just keep showing garbage. when i switch to OTAA the garbage goes away it starts printing but gets stuck on the Join request and keep on sending the join request again and again.
I hope I have explained my problem properly. Would love an input from anyone.
Thanks in advance.

matthijskooijman, 2019-10-09:
Did you use the right baudrate? I just noticed both examples use a different baudrate (see `Serial.begin()` in the `setup()` function), so that might be your problem?

v3yron, 2019-10-09:
Thanks alot for a quick reply and yes that was the issue.

matthijskooijman, 2019-10-09:
Thanks for reporting back, good to hear it's solved. I'll close this issue now :-)

closed 2019-10-09 by matthijskooijman