matthijskooijman/arduino-lmic #249: Check if send queue is empty

NullP0int, 2019-10-28:
Hi everyone,

I would like to enable the deepsleep mode from the esp32 after all packets were send so I know when it's time to enable the deepslee mode?. In this case it doesn't matter if the packets received successfully from the gateway. Is there a way to check this in a fine way?

Thanks a lot,
Wish you the best!

matthijskooijman, 2019-10-28:
There is no elegant way for this, but you can look at the `LMIC.opmode` variable. This is a bitmask of various flags, and in particular `OP_TXRXPEND` indicates whether a packet is still pending. There might be other flags that are relevant, see https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/lmic.h#L114-L129

Alternatively, you could wait for the `TXCOMPLETE` event to happen and sleep after that (either in the event handler itself, or by setting a variable and then checking that in the mainloop).

NullP0int, 2019-10-29:
Hi, thank you for your answer.

I think I can handle this but at the moment I'm facing another problem. 

I want to send multiple messages with the LMIC library in one run of my code until I'll spend my esp32 some deepsleep.

When the esp wakes up, the counter set to the value after my last send:

```c
Preferences pref;   
pref.begin(PREFERENCES_KEY, false);   
uint8_t sequenceNum = pref.getUInt("counter", 1);   
LMIC.seqnoUp = sequenceNum;
pref.end()
```

After that, I want to send my data which will received by a queue (multiple sends). I set a sendPending flag so ensure that my other send will not override the actual one. In the EV_TXCOMPLETE I set the sendPending flag to 0 if the send is complete.

```c
MessageBuffer_t sendBuffer;
  if(sendPending == 1) {
    // do nothing just wait...
    ESP_LOGD(TAG, "waiting for TXComplete");
  } else {
    if (xQueueReceive(loraSendQueue, &sendBuffer, (TickType_t)0) == pdTRUE) {
      Preferences pref;
      pref.begin(PREFERENCES_KEY, false);
      ESP_LOGD(TAG, "counter before: %d", LMIC.seqnoUp);
      sendPending = 1;
      LMIC_setTxData2(sendBuffer.port, sendBuffer.payload, sendBuffer.size, 0);
      pref.putUInt("counter", LMIC.seqnoUp);
      ESP_LOGD(LORA_TAG, "msg=\"sending packet\" port=%d loraseq=%d bytes=%d",
              sendBuffer.port, LMIC.seqnoUp, sendBuffer.size);
      pref.end();
    }
  }
```

But for some reasons the sends don't do well. Sometimes they do not appear in TTN and sometimes they are tagged with the retry flag. I disabled the frame counter checks because I think the counter don't work well. Here are some output of the logged stuff:

The first send
```
[D][lorawan.cpp:177] processSendBuffer(): counter before: 36
[D][lorawan.cpp:182] processSendBuffer(): msg="sending packet" port=1 loraseq=37 bytes=2
```

The second one
```
[D][lorawan.cpp:177] processSendBuffer(): counter before: 37
[D][lorawan.cpp:182] processSendBuffer(): msg="sending packet" port=2 loraseq=37 bytes=11
```

So I think here is a problem because the counter count up at the first send but not in the second one. So the first and the second send tagged with the same sequence number. 

Somebody got a clue why this problem appears?
Thanks!

closed 2019-10-30 by NullP0int