matthijskooijman/arduino-lmic #256: Problem joining with OTAA

tz-e, 2019-12-05:
Hello,
I am stuck at trying to join with OTAA, what I see from the serial is just this 

> 12:21:12.912 -> 625815: EV_JOINING
> 12:21:12.912 -> 625909: EV_TXSTART

I don't use TTN, instead I use [this](https://www.resiot.io/en/) service instead
From the console I see this
[image]
But there is no answer to that request, the device is seen as one registered to the network (so APPEUI and DEVEUI are right)
Do you have any suggestions?
This is the code I am mostly using.

```
void do_send(osjob_t* j) {
  if (LMIC.opmode & OP_TXRXPEND) {
    Serial.println(F("OP_TXRXPEND, not sending"));
  } else {
    LMIC_setTxData2(1, mydata, sizeof(mydata) - 1, 0);
    Serial.println(F("Packet queued"));
  }

}

static void initfunc (osjob_t* j) {
   do_send(&sendjob);
}
void setup() {
  osjob_t initjob;
  delay(1000);
  while (! Serial);
  Serial.begin(115200);
  Serial.println(F("Starting"));
  os_init();
  LMIC_reset();

  //LMIC_setClockError(MAX_CLOCK_ERROR * 10 / 100);
  LMIC.dn2Dr = SF9;
  //LMIC_setAdrMode(0);
  LMIC_setDrTxpow(DR_SF9, 14);
  
//  do_send(&sendjob);
  LMIC_startJoining();
  delay(5000);
  //LMIC_startJoining();
  os_runloop_once();
}

void loop() {
}
```

Thanks in advance

camilo-17, 2019-12-05:
Are you tried to use ABP ?

tz-e, 2019-12-06:
> 
> 
> Are you tried to use ABP ?

Yes, with ABP it works without issues
For other reasons I am trying to make OTAA work, without success

cstratton, 2019-12-20:
>Yes, with ABP it works without issues
For other reasons I am trying to make OTAA work, without success

The first thing you should determine is if the network is generating a join accept at all.  If not, it could be invalid key settings, or it could be that you are using an already used join nonce.  A proper LoRaWAN server will reject any join nonce that has previously been used.  LMiC tends to pick a random number and then slowly increment, if you have a bad RNG in your port to hardware, it may take longer and longer to get to an unused value each time.

If the join accept is being generated and transmitted by a nearby gateway, then you have to see if it is being missed due to a timing issue, ie, the node is not listening at the right time.   Are you able to receive downlinks in ABP mode?