matthijskooijman/arduino-lmic #264: Compilation error

jbrepogmailcom, 2020-01-27:
Hello, I have commented in other issue, but that has been closed, so I repeat here. I am getting following error during compiling for esp8266 and esp32:

/home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework/src/hal/hal.cpp: In function 'void hal_printf_init()':
/home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework/src/hal/hal.cpp:223:54: error: '_FDEV_SETUP_WRITE' was not declared in this scope
     fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);
                                                      ^
/home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework/src/hal/hal.cpp:223:71: error: 'fdev_setup_stream' was not declared in this scope
     fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);
                                                                       ^
Multiple libraries were found for "lmic.h"
 Used: /home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework
Multiple libraries were found for "SPI.h"
 Used: /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SPI
Using library IBM_LMIC_framework at version 1.5.0 arduino-0 in folder: /home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework 
Using library SPI at version 1.0 in folder: /home/janbenes/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SPI 
exit status 1
Error compiling for board Generic ESP8266 Module.

matthijskooijman, 2020-01-27:
You're using an older version:

> Using library IBM_LMIC_framework at version 1.5.0 arduino-0 in folder: /home/janbenes/Dropbox/Apps-180116/Arduino/libraries/IBM_LMIC_framework

The latest is 1.5.0-arduino-3, but I just remembered that the Arduino IDE mis-sorts these version numbers, so it probably installs the wrong version by default...

closed 2020-01-27 by matthijskooijman

jbrepogmailcom, 2020-01-27:
You were right, I had to manually select version in the Library manager. Latest available was arduino-2, which gives compilation error
"This example requires DISABLE_INVERT_IQ_ON_RX to be set. Update config.h in the lmic library"

Is this error fixed in version arduino-3? Thank you

matthijskooijman, 2020-01-27:
> You were right, I had to manually select version in the Library manager. Latest available was arduino-2

Then maybe the newest version is not index by Arduino yet, I have only just pushed it yesterday.

> which gives compilation error
"This example requires DISABLE_INVERT_IQ_ON_RX to be set. Update config.h in the lmic library"
> Is this error fixed in version arduino-3? I have uncommented that line in config.h, but the explanation says that uncommenting will cause that gateway is not seen but the nodes are. For LoRaWAN we want to see the gateway but not nodes, is that right?

No, this error should be resolvd by changing the config as instructed. However, the example you are using is intended for raw node-to-node LoRa communication that is completely unrelated to LoRaWAN. If you worry about gateways, I suspect you're using the wrong example.