matthijskooijman/arduino-lmic #266: Can someone help with MSB,LSB situation?

zpowellman, 2020-02-07:
I've read so many contradictory statements about this, I totally confused.  For TTN connection, are NWKSKEY, APPSKEY and DEVADDR supposed to be entered msb or lsb first?
Thanks

petzah, 2020-02-07:
I have this helper which works for me (I'm pretty sure it could be improved as I'm no proper dev)
```
const char *devEui = "0065159219FFFFFF";
const char *appEui = "FFFFFF7ED00113E2";
const char *appKey = "8FFFFF7FFE787301465910E61C9852E9";

struct LOTAA {
    byte deveui[8];
    byte appeui[8];
    byte appkey[16];
};

LOTAA lotaa;

void hexstr_to_char(unsigned char *buf, const char *hexstr, boolean lsbf = false) {
  size_t len = strlen(hexstr);
  size_t final_len = len / 2;
  if (lsbf) {
    for (size_t i = 0, j = final_len; j > 0; i += 2, j--)
      buf[j - 1] = (hexstr[i] % 32 + 9) % 25 * 16 + (hexstr[i + 1] % 32 + 9) % 25;
  }
  else {
    for (size_t i = 0, j = 0; j < final_len; i += 2, j++)
      buf[j] = (hexstr[i] % 32 + 9) % 25 * 16 + (hexstr[i + 1] % 32 + 9) % 25;
  }
}

hexstr_to_char(lotaa.deveui, devEui, true);
hexstr_to_char(lotaa.appeui, appEui, true);
hexstr_to_char(lotaa.appkey, appKey);

void os_getDevEui(u1_t* buf) { memcpy(buf, lotaa.deveui, sizeof(lotaa.deveui)); }
void os_getArtEui(u1_t* buf) { memcpy(buf, lotaa.appeui, sizeof(lotaa.appeui)); }
void os_getDevKey(u1_t* buf) { memcpy(buf, lotaa.appkey, sizeof(lotaa.appkey)); }
```

zpowellman, 2020-02-08:
Hi Peter, 

Thank for your reply.  I am interested in the Device Address, the Network Session Key and the App Session key defined in the application section of The Things Network.  My question is concerning whether these values are to be added in my LMIC based node applications in lsb or msb first format.  As follows

 

static const u1_t NWKSKEY[16] = { 0x36, 0x86, 0x7C, 0x99, 0x71, 0x06, 0x70, 0x7F, 0xAF, 0xFD, 0x2C, 0xFF, 0xED, 0x0A, 0x2E, 0xAF };

static const u1_t APPSKEY[16] = { 0x6A, 0x4B, 0xCB, 0x61, 0x4F, 0xCF, 0x5F, 0x3F, 0x9E, 0x34, 0x34, 0x71, 0xDF, 0x4F, 0xD9, 0xF8 };

static const u4_t DEVADDR = 0x13221641;

 

I don’t need a helper app, I just need clarification.  I’m not sure what you supplied, but I don’t need any reformatting.

Thanks

Dan

matthijskooijman, 2020-02-08:
IIRC all of those three must be entered MSB-first. The keys are just a pile of bytes, which do not have so much endianness, the devaddr is simply entered as a 32-bit number, so any system endianness is automatically corrected.

The tricky part is OTAA EUIs, which are 64-bit values which are specified as byte arrays, and are then sent as-is in packets, so should be *little* endian.

zpowellman, 2020-02-08:
Thanks a lot for the response.  I’m only using ABP so I guess I’m OK.

I appreciate the response,

Dan

closed 2020-02-08 by matthijskooijman