matthijskooijman/arduino-lmic #267: Library versioning

Roberto6969, 2020-02-09:
Hello,
would you be so kind and clean up mess with library versions? I have version 0 installed in Arduino IDE and it compiles. On the same computer in VSC I have version 3 installed but it does not compile due to errors as with version 0, 1 or 2.
How can I manually convince VSC it will detect the right version?
BTW: I have only version 3 files in libraries folder. 

Thank You!

matthijskooijman, 2020-02-09:
What is the mess exactly? I think the IDE might incorrectly sort and default to the -0 version, but -3 should really be the latest version and it should work.

If it does not, please offer some error messages and a proper bug report, since your current report is not really actionable :-)

Roberto6969, 2020-02-09:
Sorry, I will try to be more precise:
- i had downloaded ver 3 from Github and install it in Arduino
- nevertheless, Arduno IDE showing ver 0:
[image]
- but Arduino compiles
- library manager in VSCode shows:
[image]
- it does not compile
- in library folder is version 3 but 2 different environment showing two different version, none showing the right one

If I conclude with open points:
- Arduino IDE showing wrong version (but it compiles, so who knows what is behind the library manager)
- VSCode showing wrong number (and it not compiles, so I seems as it really has version 2)

I  checked again version in library folder. It is from 26.1.2020 - I guess this is version 3.

Please, let me know if I haven't been clear enough or if you need any additional information.