matthijskooijman/arduino-lmic #268: 'do_send' was not declared in this scope

jbrepogmailcom, 2020-02-18:
In example code ttn-abp, I get error in subject at line 

os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);

What can I do?

trlafleur, 2020-02-18:
You need to show the error, processor use, ver of ide,,,etc, etc....

charlesfayal, 2020-03-03:
Also having this issue 

`Arduino: 1.9.0-beta (Mac OS X), Board: "Discovery, Discovery L072CZ-LRWAN1, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

In file included from /Users/mac/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/arduino_lmic_hal_boards.h:21,
                 from /Users/mac/Projects/Helium/HeliumWaterMeter/quickstart/quickstart.ino:31:
/Users/mac/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/arduino_lmic_hal_configuration.h: In member function 'virtual Arduino_LMIC::HalConfiguration_t::TxPowerPolicy_t Arduino_LMIC::HalConfiguration_t::getTxPowerPolicy(Arduino_LMIC::HalConfiguration_t::TxPowerPolicy_t, int8_t, uint32_t)':
/Users/mac/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/arduino_lmic_hal_configuration.h:100:19: warning: unused parameter 'policy' [-Wunused-parameter]
  100 |   TxPowerPolicy_t policy,
      |   ~~~~~~~~~~~~~~~~^~~~~~
/Users/mac/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/arduino_lmic_hal_configuration.h:101:10: warning: unused parameter 'requestedPower' [-Wunused-parameter]
  101 |   int8_t requestedPower,
      |   ~~~~~~~^~~~~~~~~~~~~~
/Users/mac/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/arduino_lmic_hal_configuration.h:102:12: warning: unused parameter 'frequency' [-Wunused-parameter]
  102 |   uint32_t frequency
      |   ~~~~~~~~~^~~~~~~~~
/Users/mac/Projects/Helium/HeliumWaterMeter/quickstart/quickstart.ino: In function 'void onEvent(ev_t)':
quickstart:193:25: error: 'do_send' was not declared in this scope
  193 |                         do_send);
      |                         ^~~~~~~
/Users/mac/Projects/Helium/HeliumWaterMeter/quickstart/quickstart.ino: In function 'void do_send(osjob_t*)':
/Users/mac/Projects/Helium/HeliumWaterMeter/quickstart/quickstart.ino:229:23: warning: unused parameter 'j' [-Wunused-parameter]
  229 | void do_send(osjob_t *j) {
      |              ~~~~~~~~~^
exit status 1
'do_send' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

marcel151, 2020-05-23:
I also got this issue when using PlatformIO.

```
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/pro8MHzatmega328.html
PLATFORM: Atmel AVR 2.1.0 > Arduino Pro or Pro Mini ATmega328 (3.3V, 8 MHz)
HARDWARE: ATMEGA328P 8MHz, 2KB RAM, 30KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr 5.0.0 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LMIC-Arduino> 1.5.0+arduino-3
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio/build/pro8MHzatmega328/src/main.cpp.o
Compiling .pio/build/pro8MHzatmega328/libf9c/SPI/SPI.cpp.o
Compiling .pio/build/pro8MHzatmega328/lib7a4/LMIC-Arduino_ID852/aes/ideetron/AES-128_V10.cpp.o
Compiling .pio/build/pro8MHzatmega328/lib7a4/LMIC-Arduino_ID852/aes/lmic.c.o
src/main.cpp: In function 'void onEvent(ev_t)':
src/main.cpp:115:82: error: 'do_send' was not declared in this scope
             os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);
                                                                                  ^
Compiling .pio/build/pro8MHzatmega328/lib7a4/LMIC-Arduino_ID852/aes/other.c.o
Compiling .pio/build/pro8MHzatmega328/lib7a4/LMIC-Arduino_ID852/hal/hal.cpp.o
.pio/libdeps/pro8MHzatmega328/LMIC-Arduino_ID852/src/aes/ideetron/AES-128_V10.cpp:81:13: warning: 'Send_State()' declared 'static' but never defined [-Wunused-function]
 static void Send_State();
             ^
Compiling .pio/build/pro8MHzatmega328/lib7a4/LMIC-Arduino_ID852/lmic/lmic.c.o
Compiling .pio/build/pro8MHzatmega328/lib7a4/LMIC-Arduino_ID852/lmic/oslmic.c.o
*** [.pio/build/pro8MHzatmega328/src/main.cpp.o] Error 1
.pio/libdeps/pro8MHzatmega328/LMIC-Arduino_ID852/src/lmic/lmic.c: In function 'decodeFrame':
.pio/libdeps/pro8MHzatmega328/LMIC-Arduino_ID852/src/lmic/lmic.c:1040:17: warning: unused variable 'window' [-Wunused-variable]
     const char *window = (LMIC.txrxFlags & TXRX_DNW1) ? "RX1" : ((LMIC.txrxFlags & TXRX_DNW2) ? "RX2" : "Other");
                 ^
.pio/libdeps/pro8MHzatmega328/LMIC-Arduino_ID852/src/lmic/lmic.c: In function 'processJoinAccept':
.pio/libdeps/pro8MHzatmega328/LMIC-Arduino_ID852/src/lmic/lmic.c:1467:10: warning: unused variable 'mic' [-Wunused-variable]
     u4_t mic  = os_rlsbf4(&LMIC.frame[dlen-4]); // safe before modified by encrypt!
          ^
.pio/libdeps/pro8MHzatmega328/LMIC-Arduino_ID852/src/lmic/oslmic.c: In function 'os_clearCallback':
.pio/libdeps/pro8MHzatmega328/LMIC-Arduino_ID852/src/lmic/oslmic.c:54:10: warning: unused variable 'res' [-Wunused-variable]
     u1_t res = unlinkjob(&OS.scheduledjobs, job) || unlinkjob(&OS.runnablejobs, job);
          ^
```

matthijskooijman, 2020-05-24:
Normally, the Arduino IDE automatically generates forward declarations for functions, so they can be used before they are defined. In this case, I suspect this declaration is not properly generated, which prevents the `do_send` function from being found.

I see @marcel151 uses PlatformIO, I vaguely recall that PlatformIO did not implement this forward declaration generation (and thus is not 100% Arduino compatible), but I'm not entirely sure. It seems that @charlesfayal is using Arduino 1.9.x beta, which I think completely changed the way these forward declarations are generated, so this might be a bug there. Also, @charlesfayal seems to be using the MCCI LMIC version and some `quickstart.ino` sketch from (I think) elsewhere, so that problem could be completely unrelated to this repository.

As a solution, you could try adding an explicit forward declaration. I.e. add the following somewhere near the top of the sketch (e.g. just before the `onEvent` function):

    void do_send(osjob_t* j);

(this is essentially the same as the `do_send` function definition, but with the function body and its `{ }` removed, and a `;` added).

Alternatively, you can move the `do_send` function definition upwards, so it is defined before `onEvent`, which could also fix this.

If you try this, let me know if it works.

marcel151, 2020-05-24:
I saw that renaming the main.cpp to main.ino might help. And it did! It works now, but I don't know why.

matthijskooijman, 2020-05-24:
Ah, then that was the problem. The Arduino IDE (and apparently also PlatformIO), compiles `.cpp` files as-is, without any processing at all (so you get standard C++ behavior, where functions must be declared or defined before us). For `.ino` files, some preprocessing happens, such as adding forward declarations for functions.

I had noticed the `main.cpp` in the PlatformIO output, but I assumed that PlatformIO just converted all `.ino` files into a single `main.cpp` and compiled that, but apparently not. In that case, wherever you got this sketch from, it got messed up somehow (not sure how that could have ever worked, then). The examples here just `.ino` files, not `.cpp` files.

marcel151, 2020-05-24:
Thank you for the explanation. I got the sketch from here, but I just copied the contents of the "ttn-otaa.ino" to the "main.cpp" of the newly created Arduino-Project in PlatformIO.