matthijskooijman/arduino-lmic #272: Doubt on Grove RFM95 module pin mapping

Percmau, 2020-03-06:
Hi,

I am having some problems in order to connect my Grove RFM95 Lora module to the things network. The module manufacturer recommends to use the LMIC_LIBRARY to connect this Lora module to TTN. The problem is that the pin mapping of the library is not designed for the UART connector that uses the RFM95 to connect with the Grove UART port. Someone know which pin mapping should I use on the LMIC_Library code (actually using ttn-otaa.ino)? 

Thanks and sorry for the inconvenience.

trlafleur, 2020-03-06:
RFM95 connect via SPI serial and 2 I/O lines to DIO 0 and DIO 1 on the RFM95

matthijskooijman, 2020-03-06:
I'm a bit confused: Grove connectors have 2 power pins and 2 data pins. The RFM module needs SPI (4 pins) plus two DIO pins, so 6 pins in total.

Looking for the "Grove RFM95" module, I found [this one](https://www.seeedstudio.com/Grove-LoRa-Radio-868MHz.html) is that the one you're using? (for future reference: It helps to include a link with such questions, so it's clearer what you're talking about).

Looking at the [schematic for that module](https://github.com/SeeedDocument/Grove_LoRa_Radio/blob/master/res/868_sch.pdf), it seems there is an Atmega168 microcontroller on the module, so I suspect it implements some kind of UART port expander.

In short: I think bthis module is not directly supported by the LMIC library. I guess the library *could* be modified, but that won't be quite trivial.

If the module supplier recommends a particular library, that library is probably written with this module in mind, so maybe you should use that? I'm pretty sure they did not mean LMIC there, in case that's what you thought?

Percmau, 2020-03-06:
Yes, I am using [this one](https://www.seeedstudio.com/Grove-LoRa-Radio-868MHz.html).
About the library, they mention [there](https://www.seeedstudio.com/Grove-LoRa-Radio-868MHz.html), on the **Documents** field, the LMIC library. That is why I am a bit confused.

However, I will write a message to the module supplier in order to get more information about that. Maybe it is an error from the web of the module supplier. Maybe I am misunderstanding what they are saying.

Thanks for the reply, I will post something when they answer me.