matthijskooijman/arduino-lmic #273: USB error after upload sketch

skuppens, 2020-03-18:
Hi,
Every time when i upload the sketch, The COM-port gets disconnected and I get a popup stating the USB-Device isn't functioning like it should.
I hooked it up according to your explanation, but somehow I don't get it to work.

I'm trying this with a Arduino Nano 33 IoT end an RFM95 module.
Feeding the module externally 3.3V, MISO, MOSI and SCK hooked up to same pins on Arduino side. RST on pin D5, CS (NSS) on Pin D6, G0 on Pin D2, ...
This is the pin mapping of the sketch:
const lmic_pinmap lmic_pins = {
    .nss = 6,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, 4},
};

Working with Arduino IDE ver.:1.8.12 and your lmic lora library for TTN connectivity.

Already thank you for looking in to this.

matthijskooijman, 2020-03-18:
W00ps, wrong button.

This might happen when an assert happens with interrupts disabled. Assertion failures lock up in an infinite loop, and if interrupts are disabled, USB can no longer function. You could try re-enabling interrupts in the assertion handler to see if that helps (I don't have an example ready just now, let me know whether you can find this in the code).

skuppens, 2020-03-18:
I guess I found it in the hal.h file.

With these parts:

void hal_enableIRQs (void);

/*
 * put system and CPU in low-power mode, sleep until interrupt.
 */

I've read somewhere that sometimes you need to include the lowpower library as well, but I'm not sure why. Just including didn't work...

But I do not know how to handle this.

matthijskooijman, 2020-03-18:
Got a bit more time, now, sorry to plunge you into the deep before. The assert.h you mention is unrelated, LMIC implements [its own assert](https://github.com/matthijskooijman/arduino-lmic/blob/54bc51df00de18d7dd236fafac6b48e2597957f1/src/lmic/oslmic.h#L46).

The problem is in the `hal_failed()` function that handles failed assertions. It is actually a bit worse than I suggested: That function actively *disables* interrupt before looping. I should probably fix that...

For now, try removing this line:

https://github.com/matthijskooijman/arduino-lmic/blob/54bc51df00de18d7dd236fafac6b48e2597957f1/src/hal/hal.cpp#L280

skuppens, 2020-03-18:
It didn't help.
I just commented it out.

Still same problem: com port not available and a USB error stating the device isn't been recognised.
pressing the reset button 2 times gives me the ability to upload a new sketch.

matthijskooijman, 2020-03-18:
Then maybe interrupts are already enabled when the assert fails. Try putting `hal_enableIRQs();` at the start of the `hal_failed()` function (so before the error printing, since printing with interrupts disabled can also cause lockups on SAMD boards...).

skuppens, 2020-03-18:
No error now.
I can open the Serial Output window now, but nothing to see. No serial output what so ever.
So I'm not sure this is because of another error or is related to the one we're working on.

matthijskooijman, 2020-03-18:
If this removed the USB errors, then an assert is indeed triggered. I suspect you're running into the following problem, which is native USB boards do not reset when the serial port is opened, so by the time you open up the serial port, startup is complete and the error message is already gone.

Try adding `while (!Serial) /* wait */;` just after the `hal_enableIRQs();` you added. This tells the code to wait for the serial port be opened before printing the assertion failure, so then you should be able to see it.

skuppens, 2020-03-18:
20:39:53.013 -> FAILURE 
20:39:53.013 -> C:\Users\stijn\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:818

At least there is output :-)

skuppens, 2020-03-18:
at that eroor is an ASSERT.

       } else if( flags1 & IRQ_FSK1_TIMEOUT_MASK ) {
            // indicate timeout
            LMIC.dataLen = 0;
        } else {
            ASSERT(0);
        }

matthijskooijman, 2020-03-18:
Yup, so I think that means a "interrupt" line is triggered without an interrupt pending in the chip. I would suspect a problem in your pinout or wiring of the DIO pins.

skuppens, 2020-03-18:
hmmm....
I'll run this by you.
I've got on the RFM95:

Pins: VIN and GND both of them are connected to an external power (ARDUINO)
Pin: EN is not connected
Pin: G0 is connected to digital pin 2 on the Nano 33 IoT
Pin: SCK is connected to SCK on the Nano 33 IoT
Pins: MISO and MOSI are connected to MISO and MOSI on the Nano 33 IoT
Pin: CS is connected to digital pin 6 on the Nano 33 IoT
Pin: RST is connected to digital pin 5 on the Nano 33 IoT

The code I'm using is this:
`// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 6,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, 4},
};`

Not sure what I did wrong. I'll rollback the code changes we did earlier.

matthijskooijman, 2020-03-18:
Note sure what pin G0 is, is that DIO0? My [docs for the RFM95](https://cdn.sparkfun.com/assets/learn_tutorials/8/0/4/RFM95_96_97_98W.pdf) have no Gx pins?

Assuming G0 is DIO0, you configured just DIO0, but specified DIO0, DIO1 and DIO2 here:
```
    .dio = {2, 3, 4},
```
If you leave out any, you should specify `LMIC_UNUSE_PIN` instead of a random unused pin number (which might be floating and reads as high or low depending on static). Also, `DIO0` is always required and either or both of `DIO1` and `DIO2` (depending on whether you use LoRa, FSK or both). I would recommend connecting all three of them.

skuppens, 2020-03-18:
module I'm using: [Link](https://learn.adafruit.com/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts)

skuppens, 2020-03-18:
In the Hal.cpp file it states that the DIO0 AND DIO1 are required for LoRa...

skuppens, 2020-03-18:
the G0 is an IRQ pin

matthijskooijman, 2020-03-18:
Looks like Gx refer to DIOx indeed (just submitted a suggestion to Adafruit to document this).

> In the Hal.cpp file it states that the DIO0 AND DIO1 are required for LoRa...

Yup, so you need to connect at least DIO0/G0 and DIO1/G1 and specify the other one as unused, e.g.:

       .dio = {2, 3, LMIC_UNUSED_PIN},

However, I would still recommend *also* connection DIO2/G2, which is needed for FSK modulation, which can be used withing LoRaWAN as well (I think it is not used by default, but the network *could* send a command to start using it, I think).

skuppens, 2020-03-18:
Did connect them al....
Give sme this error:
21:22:34.512 -> FAILURE 
21:22:34.512 -> C:\Users\stijn\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689

matthijskooijman, 2020-03-18:
That's a different error, note the line number. What's on that line?

skuppens, 2020-03-18:
#ifdef CFG_sx1276_radio
    ASSERT(v == 0x12 );
#elif CFG_sx1272_radio
    ASSERT(v == 0x22);
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif

skuppens, 2020-03-18:
the line ASSERT(v == 0x12);

matthijskooijman, 2020-03-18:
That means the version number read is not equal to what's expected. This likely means the SPI connection is bad. Doublecheck your connections, and maybe lower SPI speed here: https://github.com/matthijskooijman/arduino-lmic/blob/54bc51df00de18d7dd236fafac6b48e2597957f1/src/hal/hal.cpp#L79

skuppens, 2020-03-18:
But SPI are just the SCK, MOSI and MISO pins.
What could he wrong there?
I'll try to lower the speed later of tomorrow.

Outlook for Android<https://aka.ms/ghei36> downloaden

matthijskooijman, 2020-03-18:
Might very well be the speed. It defaults to the max allowed for the radio, which is 10Mhz, but that's often too much for jumper wires. On 16Mhz AVR, this is less of a problem, since SPI is limited to 4Mhz IIRC, but your M0 board can probably actually do 10Mhz.

skuppens, 2020-03-19:
10:56:51.213 -> FAILURE 
10:56:51.213 -> C:\Users\stijn\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:660

This was the adjustment: static const SPISettings settings(4E6, MSBFIRST, SPI_MODE0);
instead of: static const SPISettings settings(10E6, MSBFIRST, SPI_MODE0);

matthijskooijman, 2020-03-19:
Different error again, so that's progress. What's on line 660?

skuppens, 2020-03-19:
This was the adjustment: static const SPISettings settings(4E6, MSBFIRST, SPI_MODE0);
instead of: static const SPISettings settings(10E6, MSBFIRST, SPI_MODE0);

Rock-N-RollaMann, 2020-05-05:
Hey Skuppens,

Try using an older version of the library. I was running into a similar problem with my LoRa feather m0. And that solved my problem. Hope this helps! =]

skuppens, 2020-05-07:
> 
> 
> Hey Skuppens,
> 
> Try using an older version of the library. I was running into a similar problem with my LoRa feather m0. And that solved my problem. Hope this helps! =]

I'll give it a try as soon as I find me some tinkering time again. Thanks for the tip. I'll keep you posted.