matthijskooijman/arduino-lmic #274: LMIC with other backends

Percmau, 2020-03-23:
Hello,

I know that this library works with the backend The Things Network, but does someone know if this library can work with other backends as Loriot?

I would like to use this library with my RFM95 chip in order to send the data to Loriot backend (using a configured Laird RG1186 gateway through the chip and the backend).

Thanks.

matthijskooijman, 2020-03-23:
It should implement standard LoRaWAN, so I would expect it to work (have not tried, though). The ttn-* examples are not even really TTN-specific, other than the frequency plan in the ABP example.

Dakantz, 2020-04-03:
I've been using `https://github.com/gotthardp/lorawan-server` and I've got it sending data to a backend. One of the drawbacks (or advantages, however you may see it) is the extensive configurability, which makes the setup process a little bit harder than on the TTN.
I haven't figured out how to setup downlinks properly - I can recieve some packages in RX2 but none in RX1.