matthijskooijman/arduino-lmic #278: Error with "ASSERT(v == 0x12 );"

jeremiahjohnford, 2020-05-13:
Hello 

I am currently trying to use a MKRZERO and a RFM95 module to connect to the TTN in order to periodically send and access sensor data. I have been trying to use this library for the past week and I've had some success. The library and the example code works well on the ardiuno uno which I used to first test the code (even though the uno will only stay connected for a couple messages). 

For reference Ii have been using this tutorial:
https://www.mobilefish.com/developer/lorawan/lorawan_quickguide_build_lora_node_rfm95_arduino_uno.html

Working with the MKRZERO though i have hit a road block. 
The error I keep getting is:
```
Starting
FAILURE 
C:\Users\HP\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689
```
This refers to the line 
`    ASSERT(v == 0x12 )`

From reading previous threads, I know that this "means the version number read is not equal to what's expected". I know that the SPI connection is solid and the pin definitions are accurate because I've been able to unload code using the LORA Arduino library to the MKRZERO and it runs fine. 

I've been trying to print that 'v' value, using the code that someone posted eariler:
```
#ifdef CFG_sx1276_radio
    Serial.print("Expected 0x12 from SX1276, read=0x");
    Serial.println(v,HEX);
    ASSERT(v == 0x12 );
#elif CFG_sx1272_radio
    Serial.print("Expected 0x22 from SX1272, read=0x");
    Serial.println(v,HEX);
    ASSERT(v == 0x22);
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif
```
but this doesn't work as it gives me the following error when compiling 
```
\Users\HP\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689:5: error: 'Serial' undeclared (first use in this function)

     Serial.print("Expected 0x12 from SX1276, read=0x");
```
I've tried about everything that I can think of up until this point.

Thanks for the help in advance!

matthijskooijman, 2020-05-13:
> I've been trying to print that 'v' value, using the code that someone posted eariler:

This lives in a .c file, so you cannot use `Serial` (which is a C++ object) there.

Instead, you can use e.g. `printf("V=0x%x\n", v);` to print the value, provided you uncomment this line: https://github.com/matthijskooijman/arduino-lmic/blob/54bc51df00de18d7dd236fafac6b48e2597957f1/src/lmic/config.h#L33

montanarco, 2020-08-25:
Hi 
Currently I’m experimenting the same issue, I connected an Arduino UNO to a Ra-02(sx1278), I’m using LMIC library I defined this frequency **#define CFG_us915** 1 and this board **#define CFG_sx1276_radio 1**  then I got an error in this line libraries\arduino-lmic-master\src\lmic\radio.c:706 When I check and I found this: 
#ifdef CFG_sx1276_radio
    ASSERT(v == 0x12 );

So I added a couple of printf as suggested as debugging method:
#ifdef CFG_sx1276_radio
    printf("Expected 0x12 from SX1276, read=0x");
    printf("V=0x%x\n", v);
    ASSERT(v == 0x12 );

The output was
  22:03:59.294 -> Starting
  22:04:00.852 -> **Expected 0x12 from SX1276, read=0xV=0x0**
  22:04:00.886 -> FAILURE  

That help me to realized that v is **0x0** so is not going anywhere, but I wonder if is a matter of how the board is linked 
I’m using this schema: https://www.mobilefish.com/images/developer/lorawan_rfm95_arduino_large.jpg 
I double check it and it is exactly the same, Here is my pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, LMIC_UNUSED_PIN},
};

Do you think my Arduino got damaged? Maybe my Ra-02 module? or is there anything I’m missing?

Any help is going to be highly appreciated, thanks a lot!
thanks to @matthijskooijman great library indeed!

hallard, 2020-08-25:
If you’re using this wiring I would suggest reading the comment of Ivan on the original site.
You’re driving module with 5V logic level for some even out of specs it works other not. Looks like it’s your case 
I suggest you to use 3v3 arduino Mini board 
This PCB has even all wired to rfm95 module 
https://github.com/hallard/Mini-LoRa

guljanjua, 2021-09-17:
Hi,

I hope you are doing good. Please replace this:

`ASSERT(v == 0x12 );`

with this:

`if(v != 0x12 )
        return 0;`

matthijskooijman, 2021-09-17:
@guljanjua If this assert triggers, that's a symptom of many different problems, so there's no single cause here. The fix you suggest will, I think, simply skip initialization of the chip if it doesn't look like the right chip (and/or communication is not working), which is not really a fix (I would doubt that the library would actually *work* after that). If your suggestion is to replaces this so the error can be handled elsewhere rather than doing a fatal assert, then that would make some sense, but this library is no longer being actively developed. I would then suggest looking at the MCCI LMIC version (see the README), which is still being developed.