matthijskooijman/arduino-lmic #279: Used arduino-Lmic with board ATmega

TuongBui16, 2020-05-14:
I have problems using the Lmic library on the ATmega board "Error compiling for board ATmega1284." How should I use this library on my board or Lmic does not support my ATmega board?

matthijskooijman, 2020-05-14:
I think it should support that chip, I've used it succesfully on a mega2560 as well. Maybe you could share the error messages you are getting?

Alternatively, you could consider trying https://github.com/mcci-catena/arduino-lmic a fork of this library that is a bit more actively maintained and has more documentation.

Roberto6969, 2020-05-14:
I have LMIC running on 2560 Mega Pro and 2560 Mega Core - works fine, also in deep sleep.

TuongBui16, 2020-05-14:
> Tôi có LMIC chạy trên 2560 Mega Pro và 2560 Mega Core - hoạt động tốt, cũng trong giấc ngủ sâu.

Hello, I would be very happy if you could share the document, the code you are using with everyone. I am in need of anything related to my ATmega 1284. Thank you