matthijskooijman/arduino-lmic #280: radio.c codes

Roberto6969, 2020-05-14:
Hello Matthijs,

maybe it is a banal question but anyway: are radio codes listed with description somewhere?

Just for an example:
Is radio.c code 689 mentioned here:
C:\Users\HP\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689
explained what it means?

We don't want to disturb you anytime when we discover new code ;-)

Thank You!

matthijskooijman, 2020-05-14:
It's somewhat documented: If you open up radio.c and go to line 689, you'll see exactly where the assertion failure was generated and what expression was false. Then what that means exactly, should be deduced from the code around it.

Roberto6969, 2020-05-14:
Thank you - it makes sense.

closed 2020-05-14 by Roberto6969