matthijskooijman/arduino-lmic #281: Lora Node unable to receive downlink messages

sgodey, 2020-05-23:
I am using RFM95W module along with ATMega328p as an IoT node. I have a RAK2245 based gateway that is continuously sending packets at 923.3 and SF12. Below is the log output. I am dumping RSSI when ever OpMode changes. All times, I am seeing RXTIMEOut. I have been stuck at this problem for long time. Any help is greatly appreciated.
  
Starting
RXMODE_RSSI
Packet queued
1182: Opmode: 0x8b 
1220: rssi: -164 dBm MdmStat: 0x10 
26841: Opmode: 0x89 
26882: rssi: -164 dBm MdmStat: 0x10 
26989: Opmode: 0x88 
27053: rssi: -164 dBm MdmStat: 0x0 
83443: RXMODE_SINGLE, freq=923300000, SF=10, BW=500, CR=4/5, IH=0
83459: Opmode: 0x8e 
83570: rssi: -44 dBm MdmStat: 0x4 
96100: Opmode: 0x89 
96142: rssi: -45 dBm MdmStat: 0x14 
96245: Opmode: 0x88 
96305: rssi: -164 dBm MdmStat: 0x0 
140438: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
140457: Opmode: 0x8e 
140578: rssi: -41 dBm MdmStat: 0x4 
159538: Opmode: 0x88 
159582: rssi: -164 dBm MdmStat: 0x0 
159662: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued

matthijskooijman, 2020-05-24:
Did you see and try https://github.com/matthijskooijman/arduino-lmic#problems-with-downlink-and-ota ?

sgodey, 2020-05-24:
I am using 10% clock correction.

LMIC_setClockError(MAX_CLOCK_ERROR * 10 / 100); 

And my RX2 settings are 923.3 MHz, BW500 and SF12. I am using **util_tx_continous** from **https://github.comLora-net/lora-gateway**, to generate a continuous stream of Lora packets at Freq-923.3, BW-500, SF12.

yangkangyk, 2020-05-30:
Does it work? I also have same problems.

sgodey, 2020-05-30:
> Does it work? I also have same problems.

It worked for me. My problem was ...
1. Not properly aligning the channel frequencies/SF/BW between transmitter and receiver.
2. The packet format between tx and rx was little bit different (receiver was expecting explicit header and transmitter was sending implicit header)

Once I fixed those and use the LMIC_setClockError, it worked for me.

mrx23dot, 2022-03-06:
Please close issue.