matthijskooijman/arduino-lmic #285: Error if DIO1 is not on SPI SS pin (Esp32 + RFM95C)

kevin192291, 2020-06-16:
I have an RFM95C connected to an ESP32 and running the arduino-lmic library.
Everything works great if I use the pin map:
```
const lmic_pinmap lmic_pins = {
    .nss = 5,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 27,
    .dio = {2, 15, 4},
};
```
Note that .dio[1] is mapped to GPIO15 on my ESP32, this is my SPI SS pin. If i were to move dio1 down just one pin to GPIO8 a standard GPIO pin that is not being used. I get a boot error and loop.
I would really really love to free up that SPI SS pin, any ideas as to how to work with this or if not, why the RFM95C requires 2 SPI pins.

The docs do not say anything about requiring GPIO1 to be on a spi port, and dio2 doesn't matter where it is mapped.

Any Ideas?

trlafleur, 2020-06-16:
Check the data sheet on the ESP32 to make sure it can be used as an input.
  There are some restrictions.  Also a quick program to read the pin and
toggle the led can tell you a lot

matthijskooijman, 2020-06-16:
I'm pretty sure this is not something dictated by LMIC, it just needs any I/O pin capable of being used as an input. So as suggested, this is probably an ESP limitation that I can't really help you with, so I'm going to close this. Feel free to continue discussing here, if others do have suggestions.

closed 2020-06-16 by matthijskooijman