matthijskooijman/arduino-lmic #286: Date to RTC

Samuel-ZDM, 2020-07-09:
Hello!

I am trying to get the gateway date to configure the RTC.
Is there any function that I can do this for?
Or can I only do this by inserting the value in the downlink field in the TTN, for example?

matthijskooijman, 2020-07-16:
Normally, the gateway date is not known by the node, so there is no trivial way to  obtain it. I think there is a "request time from server" command defined in LoRaWAN (though that might be LoRaWAN 1.1). I'm not sure if this is somehow already supported by LMIC and how you would get it (and no time to check now).

trlafleur, 2020-07-16:
I set up node-red to send the time ever 1, 10 min and 1hr, to MQTT, and
have a task to send it Periodically to Lora nodes that request it... for
most nodes, they make a request once a day, and at start up...

Samuel-ZDM, 2020-07-22:
Thank you.

I was wondering if LMIC could do that.
So, the responsible for this is the application server.

closed 2020-07-22 by Samuel-ZDM