matthijskooijman/arduino-lmic #288: Can't get downlink in MSP430 port

antonio-emilio, 2020-07-21:
I've made a port of this LMIC library to the MSP430. I've made the uplink works perfectly, but I can't get the downlink. I'm using the ttn-abp example. I already tested adding LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100), but didn't worked...

In this port, I couldn't make the osjob work, so, after sending the package, I call the funciton setupRx2DnData(). Am I correct?
I'm kinda lost in this downlink thing... I've trying to make this work for about 2 weeks...

matthijskooijman, 2020-07-21:
I'm doubtful I can help you with this, certainly with so little information.

> In this port, I couldn't make the osjob work,

That seems problematic, since the osjobs are used for various timing-related jobs. `setupRx2DnData()` is  called to set up the RX2 window, but before that another function sets up the RX1 window I think. Both should be called at pretty much exactly the right moments, and there's probably more osjobs that you're missing. I would highly suggest making the osjob stuff work (I'm surprised it doesn't just work, since it's just fairly plain C code with a linked list of jobs to poll, nothing fancy or platform-specific in there).

DeuxVis, 2020-07-21:
@antonio-emilio For your information I did a port of Arduino-LMIC to Energia - the arduino IDE for MSP430 dev boards.

Haven't used / updated it recently but here it is : https://github.com/DeuxVis/energia-lmic

Hope this may be helping, forgive me for the interruption if not.

mirkomatonti, 2022-07-20:
@antonio-emilio Hi I'm having the same issue. Do you mind sharing the your HAL implementation? Thanks!