matthijskooijman/arduino-lmic #289: Communication between node and gateway

Samuel-ZDM, 2020-07-22:
Hello
Node is able to know if it was able to communicate with the gateway?

I would like to know if the node was able to create the session between it and the gateway.

matthijskooijman, 2020-07-22:
LMIC works with the LoRaWAN protocol, which establishes a session between the node and a backend server, through a gateway. I suggest you read up on that protocol to figure out how this works. Your question is a bit vague, you'll have to be more specific if you need further info.

I'm going to close this issue, since it is not a bug or feature request (but you can still post more comments).

closed 2020-07-22 by matthijskooijman

Samuel-ZDM, 2020-07-28:
Thanks for the feedback.

My idea is to know if the final device managed to create a communication with any gateway.
I am wanting to do this, because I want to try to implement a LoRaWAN network with hops. But from what I have studied so far, the LMIC protocol does not support dual communication, between nodes together with communication with the gateway.