matthijskooijman/arduino-lmic #293: EV_TXCOMPLETE taking time to occur : Arduino Pro mini + LoRa RFM95W

fguiet, 2020-09-20:
Hi,

__Here is my configuration:__

* Library used : https://github.com/matthijskooijman/arduino-lmic/releases/tag/1.5.0%2Barduino-3
* Arduino Pro Mini 8Mhz, 3.3v
* LoRa RFM95W

__Use case__

* Watermeter system to check my water consumption.

__Issue__

When no water is consumed, my module sends a LoRa message every hour including battery voltage...

As you can see on the log below, the first message occurs one hour after the module has started.
Ok, everything fine here ... EV_TXCOMPLETE occurs 2sec after paket queued which seems to be ok.
But...starting from the second message....you can see that the EV_TXCOMPLETE event occurs 12min45s after packet has been queued ...
Even wosrt on the third message...EV_TXCOMPLETE event occurs after 23min from paket queued ...
Then on the 4th...everything's fine....

Can someone have an explanation ? and a workaround ?

I forgot to mention that I did not see any problems when I am consuming water ... because (I thing) messages are sent very often...The problem seems to occur when no message has been sent for a long time (here one hour).

2 things I have already tried :

1. Before calling : LMIC_setTxData2

LMIC.bands[BAND_MILLI].avail = os_getTime();
LMIC.bands[BAND_CENTI].avail = os_getTime();
LMIC.bands[BAND_DECI].avail = os_getTime();

2. During setup 

LMIC_setClockError(MAX_CLOCK_ERROR * 10 / 100); 

Here a log:

20:43:42.452 -> Setup completed, starting...
21:43:48.086 -> Sending battery voltage info
21:43:48.086 -> Sending message
21:43:48.086 -> Time is : 3600005
21:43:48.182 -> Packet queued
21:43:50.490 -> 225150989: EV_TXCOMPLETE (includes waiting for RX windows)
21:43:50.490 -> Time is : 3602421

22:43:54.173 -> Sending battery voltage info
22:43:54.173 -> Sending message
22:43:54.173 -> Time is : 7200006
22:43:54.268 -> Packet queued
22:55:39.098 -> 225547350: EV_TXCOMPLETE (includes waiting for RX windows)
22:55:39.098 -> Time is : 7903731

23:44:00.286 -> Sending battery voltage info
23:44:00.286 -> Sending message
23:44:00.286 -> Time is : 10800007
23:44:00.380 -> Packet queued
00:07:27.680 -> 225942112: EV_TXCOMPLETE (includes waiting for RX windows)
00:07:27.680 -> Time is : 12205015

00:44:06.439 -> Sending battery voltage info
00:44:06.439 -> Sending message
00:44:06.439 -> Time is : 14400008
00:44:06.533 -> Packet queued
00:44:08.810 -> 363280267: EV_TXCOMPLETE (includes waiting for RX windows)
00:44:08.857 -> Time is : 14402424

fguiet, 2020-09-26:
Hiya there,

Quick update on this issue:

Recall : When no water is consumed the sensor is sending a message everyhour (with the bug described on this issue)...I reduced this interval to 30 minutes....and everything works like a charm.

My 2 cents,
Fred

simortus, 2021-05-04:
You can write to the reg and change the opmode. The TX does not happen because there buffer is still busy or not entirely freed after the last TX event. buggy code i guess. 
You can debug the LMIC.opmode and check if its not at OP_TXRXPEND.

 Serial.println(F("OP_TXRXPEND, not sending"));

this is how i got rid of it 
        os_radio(OP_TXDATA);
        LMIC.opmode = OP_TXDATA;
       LMIC_setTxData2(1, mydata, sizeof(mydata)-1, 0); 

hope it helps.