matthijskooijman/arduino-lmic #295: PIN MAPPING PROBLEM

K3RN3L97, 2020-09-23:
So, I have a bare RF95 Module with the following schematic 

[image]

and am using an arduino UNO as well with the following pinouts 

[image]

and this is my pin mapping 

`const lmic_pinmap lmic_pins = {
  .nss = 10,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 9,
  .dio = {2, 6, 7},
};`

am always stuck at joining, do you think the mapping is the problem? thanks in advance

descartes, 2020-09-25:
Arduino Uno = 5V device. RFM95 = 3.3V device. Try an Arduino Pro Mini.

K3RN3L97, 2020-09-28:
it's not a problem, all i have to do is connect it to the 3.3v pin on the 328p. Am just asking about the pin mapping for the 328p or the uno rather than the mini
Thank you

simortus, 2021-05-04:
if the mapping is the problem you will be stopped at the running time with an error.
This is more a timing error, where your chip is not at the right OPMODE or could be your gateway settings

hallard, 2023-06-01:
@Phairou18 if you look at the first warning of the readme of this repo you can see it's no longer maintained and suggest to use MCCI library instead, that is more up to to date and with an 2560 would perfectly fits into.

I would not expect any support from here.

Phairou18, 2023-06-01:
@hillard I hadn't noticed I'm new here.
Do you have a link that explains how to use the MCCI library?

hallard, 2023-06-01:
A good start is documentation on the according MCCI LMIC repository.
https://github.com/mcci-catena/arduino-lmic

Phairou18, 2023-06-01:
OK, thank you @hallard 
I'm gonna check it rightnow.