matthijskooijman/arduino-lmic #296: IBM LMIC framework library version 1.5.0 + arduino-3 is not available in my library manager

Erik84750, 2020-09-27:
This version (arduino-3) does not show up in my library manager for the IBM LMIC framework library. Currently version ardiuno-2 is in use.

Also the current example for ttn-otaa (2018) has syntax errors on line 48 and 49: the "//" comment characters are replaced by "#".

When corrected there still is compilation error on line 117: 'LMIC_getSessionKeys' was not declared in this scope

matthijskooijman, 2020-09-27:
> This version (arduino-3) does not show up in my library manager for the IBM LMIC framework library. Currently version ardiuno-2 is in use.

Yeah, I saw that before. I reported https://github.com/arduino/Arduino/issues/10158 about this a while ago but now I see I missed the replies there. Now I know the problem, so I'll try to get a new version in the library manager again.

> Also the current example for ttn-otaa (2018) has syntax errors on line 48 and 49: the "//" comment characters are replaced by "#".

Which version is that? I don't see that here https://github.com/matthijskooijman/arduino-lmic/blob/cace030591ffec6314736467ff6bad6e0350c4fd/examples/ttn-otaa/ttn-otaa.ino#L48-L49

> When corrected there still is compilation error on line 117: 'LMIC_getSessionKeys' was not declared in this scope

That's weird. Are you sure you've installed a proper copy of this library? It sounds like you have something different, or maybe some corrupted versions or something?

In any case, you could maybe try the MCCI LMIC library, which is further developed and is available through the library manager. I haven't really tried it yet, but I expect it will be better in a lot of areas.

Erik84750, 2020-09-27:
The line numbers you show correspond with lines 64 and 65 here; the program shows on line 3: Copyright 2018 Terry Moore, MCCI (line 2 copyrights 2015 to you and Thomas Telkamp).

The compilation error refers to the MCCI_LoRaWAN_LMIC library (which hence is in use). I suspect I may have to replace FILLMEIN with a value, which I didn't?

matthijskooijman, 2020-09-27:
The MCCI_LoRaWAN_LMIC is not developed here, and I have little experience with it. I suggest you look for support at their repo: https://github.com/mcci-catena/arduino-lmic

closed 2020-09-27 by matthijskooijman