matthijskooijman/arduino-lmic #297: :warning: This repository is retired, alternatives recommended

matthijskooijman, 2020-10-03:
This repository is no longer maintained. The recommended alternative is the [MCCI version of LMIC](https://github.com/mcci-catena/arduino-lmic), which is based on this one, but has seen much improvements over the years and much better documentation. If you need support for the SX126x, you can consider [Lacuna's port of BasicMAC](https://github.com/LacunaSpace/basicmac), which is based on a parallel development based on the original LMIC.  Only when you have very tight constraints on RAM or flash should this version still be considered.

---

Over the years, this version of the LMIC library has not received the attention and time from me that it deserved. I'm afraid that my neglect of this repository and the resulting fragmentation and unclarity around LMIC has not been helpful for the LMIC and LoRaWAN / Arduino community and ecosystem as a whole. I'm sorry that people have run into problems with their projects without proper response, or have invested time in providing pull requests that have not been merged here. However, I hope that taking this step will provide the clarity and focus for users and developers and allows the ecosystem to become stronger again.

To better understand how and why things have happened as they did, let me walk you through some history.

In 2015, I needed to do some LoRa experiments in an Arduino environment, so I looked for existing code that could talk to the SX127x radios. I wasn't particularly interested in LoRaWAN yet at the time, but LMIC seemed promising and easy to port to Arduino, so I did just that. A few months later, The Things Network launched and I also had a usecase of my own that could make good use of LoRaWAN, so I polished up things to allow using LoRaWAN with The Things Network and a lot of people started using it for this as well.

However, to make the library work on Arduino, I had selectively imported files from the upstream LMIC repository, and moved files around to accomodate Arduino, losing the ability to use this repository outside of Arduino. This meant that importing changes from the upstream LMIC repository was hard and that any improvements made to the core code in this repository would not benefit any other platforms. For this reason, I was immediately quite hesitant with making signficant changes, to this repo to prevent deviating from the upstream version to the point of changes being unmergable between both. The fact that upstream development happened privately, only publishing zips of release versions, also complicated things.

A year or so later, IBM stopped developing LMIC and released their last 1.6 release (under a more liberal license, on my request). For me, this was the incentive to restart LMIC-for-Arduino from scratch, based on the 1.6 version and new license and this time keeping the original file layout and build system intact and *adding* Arduino support on top of that. My intention was to try and make this the standard LMIC version (by this time, a few people had already forked my repo and applied some improvements they needed), hosted under a github organisation where more people could collaborate on the library. I started [this new repo](https://github.com/lmic-lib/lmic) in 2016, ported over more changes from this repo in 2017 and was *almost* ready to announce it as a rally point for all LMIC development, but for various reasons (mostly a chronic surplus of other projects to work on, but also as the announcement of the LoRaWANMiniMouse and BasicMAC LoRaWAN stacks that cast some doubt on the future of LMIC), I never got around to do so unfortunately (which, looking back, seems like a bad choice, but well, here we are).

Early 2019, the [BasicMAC LoRaWAN stack](https://github.com/lorabasics/basicmac) was also published, which is a privately developed continuation of LMIC, which AFAIU was later acquired and released by Semtech. Initially, this seemed like it could maybe serve as a better starting point for renewed LMIC-based development. Conveniently, Lacuna was in need of a LoRaWAN stack that supported the newer SX126x chip series and was willing to sponsor [porting BasicMAC to Arduino](https://github.com/LacunaSpace/basicmac), which I undertook late 2019 and early 2020. This time, I used the repository setup and scripts I had created for the "new" version of LMIC before, to allow the Arduino support to co-exist with the existing platforms and build system. However, since then Semtech has again stopped BasicMAC development, there is still a lot of things in BasicMAC that would need to be further improved, I found that the memory usage is significantly larger (too big for 328p as it is now) and porting over the improvements separately made to LMIC over the years (especially in the MCCI version) would be quite a large, if not impossible task. There does seem to be one [other active fork by one of the Semtech developers](https://github.com/mkuyper/basicmac), but that already deviates from the Lacuna version. Maybe someone will still pick up the work of unifying all these developments, or maybe some of the BasicMAC improvements can be ported back to LMIC instead, but realistically I will not be the one to pull that cart.

In the meanwhile, I mostly neglected this repository, leaving pull requests unreviewed and unmerged and leaving support requests unanswered most of the time. I didn't want to mark this repository as officially unmaintained until I had the new repo published as an alternative to recommend, but that never happened.

Over the years, the [MCCI fork of this repo](https://github.com/mcci-catena/arduino-lmic) did see additional development, including some significant structural changes, making my original plan of restarting development with a clean repo less and less feasible. But keeping this repository in this "limbo" state of maintenance is also bad, so the next best plan seemed to just retire my repository and recommend people to use the MCCI version instead. But I didn't want to do that blindly, without trying the MCCI version in my own projects, which I also did not get around to in the last months.

Recently I finally gave the MCCI version a whirl and found it looks very good. Documentation is a lot more complete, it contains a lot of extra fixes and features and still works pretty much as a drop-in replacement. I did find that it uses a bit more flash and RAM (which makes sense given extra features), but still small enough to run in our 328p-based project. For very-resource-constrained projects, maybe this LMIC version can still be useful, but in general, I would really recommend people to go with the MCCI version instead. There are still a few small functional improvements made to this repository after the MCCI version was forked, which I plan to review and submit to the MCCI repository where appropriate.

Oliv4945, 2020-10-03:
Thanks for all the work you initially did that kick-started lots of projects!

Samuel-ZDM, 2020-10-08:
Thank you for your work. You can be sure that you have helped many people.

plachta-zabatt, 2020-10-13:
Thank you for work!