matthijskooijman/arduino-lmic #298: Dragino Lora Shield Error in radio.c after trying connetcting to TTN

MoShawa, 2020-12-09:
Hi there, 
I am working on  Lora Iot development kit v 2 from Dragino in 868 MHz. I am implementing Lmic master for sendong Temp & Hum. to TTN using otaa. So,  my single channel gateway is LG01. So , I got an error in radio.c:441 
I also modified the freqs' in  config.c file  for matching my region's frequency, which are:

`#define CFG_eu868 1
//#define CFG_us915 1
//#define CFG_au921 1
//#define CFG_as923 1
//#define CFG_in866 1

#define LG02_LG01 1

//US915: DR_SF10=0, DR_SF9=1, DR_SF8=2, DR_SF7=3, DR_SF8C=4 
//       DR_SF12CR=8, DR_SF11CR=9, DR_SF10CR=10, DR_SF9CR=11, DR_SF8CR=12, DR_SF7CR
#if defined(CFG_us915) && defined(LG02_LG01)
// CFG_us915 || CFG_as923 
#define LG02_UPFREQ   902320000
#define LG02_DNWFREQ  923300000
#define LG02_RXSF     3     // DR_SF7  For LG01/LG02 Tx
#define LG02_TXSF     8      // DR_SF12CR For LG02/LG02 Rx
#elif defined(CFG_eu868) && defined(LG02_LG01)
// CFG_eu868
//EU868: DR_SF12=0, DR_SF11=1, DR_SF10=2, DR_SF9=3, DR_SF8=4, DR_SF7=5, DR_SF7B=1, DR_FSK, DR_NONE
#define LG02_UPFREQ   868100000
#define LG02_DNWFREQ  868100000
#define LG02_RXSF     7      // DR_SF7 For LG01/LG02 Tx
#define LG02_TXSF     12     // DR_SF12 For LG02/LG02 Rx
#endif
`

I am following the user manual from dragino. 

Any help ?

this code in radio.c file from 411 to 459 
`static void configPower () {
#ifdef CFG_sx1276_radio
    // no boost used for now
    s1_t pw = (s1_t)LMIC.txpow;
    if(pw >= 17) {
        pw = 15;
    } else if(pw < 2) {
        pw = 2;
    }
    // check board type for BOOST pin
    writeReg(RegPaConfig, (u1_t)(0x80|(pw&0xf)));
    writeReg(RegPaDac, readReg(RegPaDac)|0x4);

#elif CFG_sx1272_radio
    // set PA config (2-17 dBm using PA_BOOST)
    s1_t pw = (s1_t)LMIC.txpow;
    if(pw > 17) {
        pw = 17;
    } else if(pw < 2) {
        pw = 2;
    }
    writeReg(RegPaConfig, (u1_t)(0x80|(pw-2)));
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif /* CFG_sx1272_radio */
}

static void txfsk () {
    // select FSK modem (from sleep mode)
    writeReg(RegOpMode, 0x10); // FSK, BT=0.5
    ASSERT(readReg(RegOpMode) == 0x10);
    // enter standby mode (required for FIFO loading))
    opmode(OPMODE_STANDBY);
    // set bitrate
    writeReg(FSKRegBitrateMsb, 0x02); // 50kbps
    writeReg(FSKRegBitrateLsb, 0x80);
    // set frequency deviation
    writeReg(FSKRegFdevMsb, 0x01); // +/- 25kHz
    writeReg(FSKRegFdevLsb, 0x99);
    // frame and packet handler settings
    writeReg(FSKRegPreambleMsb, 0x00);
    writeReg(FSKRegPreambleLsb, 0x05);
    writeReg(FSKRegSyncConfig, 0x12);
    writeReg(FSKRegPacketConfig1, 0xD0);
    writeReg(FSKRegPacketConfig2, 0x40);
    writeReg(FSKRegSyncValue1, 0xC1);
    writeReg(FSKRegSyncValue2, 0x94);
    writeReg(FSKRegSyncValue3, 0xC1);
    // configure frequency`

matthijskooijman, 2020-12-09:
Did you see hat I recommend switching the to MCCI version of LMIC, rather than this one?

> So, my single channel gateway is LG01.

I have no experience with single-channel gateways, I'm not sure what you need to change in LMIC to support those (also, remember that single-channel gateways might be nice for testing, they do not really scale well).

> So , I got an error in radio.c:441

If it's a compile-time error, what error? If it is a runtime assertion failure, what is line 441 exactly?

MoShawa, 2020-12-09:
Hello, 
thank you for your response. 

[image]
[image]

matthijskooijman, 2020-12-09:
Hm, it's weird that it ends up in `txfsk`, since I don't think FSK is ever used during join... I wonder if your changes to the frequencies might have also changed the DR definitions somehow causing it to use FSK. It's also perfectly possible that FSK doesn't actually work, since it's rarely or never used in practice.

Looking more closely at your code, you have these `LG02_` defines, which I don't recognize. I suspect you're not actually using this version of LMIC, but some modified (by Dragino maybe) version. I suggest you either:
 - Seek support from whomever supplied this LMIC library version
 - Try the MCCI LMIC version (see the README in this repo) instead

MoShawa, 2020-12-10:
> I wonder if your changes to the frequencies might have also changed the DR definitions somehow causing it to use FSK.
Yes I changed the frequencies according to user manual.
from user manual :
[image]

my parameters' changes 
[image]
[image]
[image]

matthijskooijman, 2020-12-10:
Right, but my point is that the version of LMIC you *start with*, before your modifications is not *this version*. e.g. see https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/config.h which does not have these `LG02_*` macros. So I suspect you're using some version modified by Dragino or someone else, which I cannot really provide support for.

simortus, 2022-01-10:
radio.init failures are most related to pin mapping! 
So check your pin mapping and get back to me :)
I used dragino with this repo and the dragino modified one. Both work perfectly if you set up the pins correctly. 
RST = 9
SS = 10
DIO = {2,6,7} //  dio0 , dio1 and dio2 respectively.
good luck